Ai-Thinker-Open/ESP32-A1S-AudioKit #1: SPI

packedbox, 2020-03-30:
Hi I'm trying to control a RC522 NFC Reader with the SPI bus on the device, but I can't figure out how to make it work. 

As a test, I use the DumpInfo sample from https://github.com/miguelbalboa/rfid in the Arduino IDE

On another basic esp32 device ( HELTEC OLED ESP32 )  using GPIOS 18 19 21 22 23, it works perfectly but not on the ESP32-A1S-AudioKit. 

Any Idea what can be wrong ? 
Thanks, 
Nicolas.

xuhongv, 2020-03-31:
- Please tell me how youconnect board and RC522 NFC？

packedbox, 2020-04-03:
Hi thanks for your answer, I use the Audio KitBoard

The wiring is the following :
RC522     ESP32
----------------
CLK       ->    18
MISO     ->   19
CS/SDA ->   21
RST       ->   22
MOSI    ->   23

And the sample sketch i use to test ( working fine with an another ESP and the same wiring ) : 

#include <SPI.h>
#include <MFRC522.h>

#define RST_PIN        22          // Configurable, see typical pin layout above
#define SS_PIN          21         // Configurable, see typical pin layout above

MFRC522 mfrc522(SS_PIN, RST_PIN);  // Create MFRC522 instance

void setup() {
	Serial.begin(115200);		// Initialize serial communications with the PC
	while (!Serial);		// Do nothing if no serial port is opened (added for Arduinos based on ATMEGA32U4)
	SPI.begin();			// Init SPI bus
	mfrc522.PCD_Init();		// Init MFRC522
	delay(250);				// Optional delay. Some board do need more time after init to be ready, see Readme
	mfrc522.PCD_DumpVersionToSerial();	// Show details of PCD - MFRC522 Card Reader details
	Serial.println(F("Scan PICC to see UID, SAK, type, and data blocks..."));
}

void loop() {
	// Reset the loop if no new card present on the sensor/reader. This saves the entire process when idle.
	if ( ! mfrc522.PICC_IsNewCardPresent()) {
		return;
	}

	// Select one of the cards
	if ( ! mfrc522.PICC_ReadCardSerial()) {
		return;
	}

	// Dump debug info about the card; PICC_HaltA() is automatically called
	mfrc522.PICC_DumpToSerial(&(mfrc522.uid));
}`

packedbox, 2020-04-22:
Hi Managed to connect a NFC reader using I2C but the I2C board are more expensive and usualy harder to find, so I'm still interested to know how to make a SPI MFRC522 work with this board. Thanks.

schreibfaul1, 2020-05-12:
C58 ... C61 (100nF) prevent GPIOs 5, 18, 19 and 23 from being used for SPI.  Alternatives are GPIOs 12 ... 14 (JT_MTxx) on P1

Pewibe, 2020-10-19:
Reason of my comment today is the same problem and this still open issue.
@xuhongv You requested information, you got the information, but you didn't answer. Any comment today?
@packedbox What is the status concerning RC522 today? The board V2.2 uses 
SD             ESP32-GPIO
CLK           14
MISO        2
MOSI        15
CS/DATA3 13
So CLK, MISO, MOSI for the RC522 have to be the same pins. Only CS/SDA have to be a new one like 12 or 19 and additionally RST for the RC522 by using 21 or 22.

stetro, 2021-02-12:
Hey, I had the same issue - I wanted to run the RC522 via SPI or I2c and I also wanted to have a couple of spare IO Ports. Unfortunatly I could not manage to have a reliable SPI setup and also using the SD card continuously and therefore bought another PN532. It supports I2C (configurable with switches on most PCBs) with IRQ interrupts. Works great with the following Setup:

SDA 22
SCL 13
IRQ 19

I used this idf port by @lucafaccin https://github.com/lucafaccin/esp-pn532 and lowered the I2C clock rate to 10000. 

Maybe this comes handy for someone with similar issues.

marcel-licence, 2021-07-03:
Hi, a little mod can help to get the pins free for spi and also the buttons running: https://youtu.be/r0af0DB1R68 
I am using following pins:
MOSI 0 
SCLK 19
But other pins may also work

ManosDim, 2023-03-05:
> Hi, a little mod can help to get the pins free for spi and also the buttons running: https://youtu.be/r0af0DB1R68 I am using following pins: MOSI 0 SCLK 19 But other pins may also work The mod is required because there are capacitors on the audio kit causing problems

Hey, for me the info here is kinda confusing. I watched your video and the IOs that you made free by the hack are 13 19 23 18 5, but you are using the IOs 0 mosi and 19 sclk for the SPI. Also in the documentation here https://github.com/pschatzmann/arduino-audiokit/wiki/GPIO-Overview-by-Selected-Board , the available IOs are 13 cs, 2 miso, 15 mosi and 14 clk. Is it possible to explain to me the exact wiring that you did in the pins on the board ( I cant find the 0 pin in the header) and tell me where i can track the miso pin? Thanks

marcel-licence, 2023-03-05:
On my Pin header I have:
GND IO0 RST TX0 RX0 3V3 3V3
IO21 IO22 IO19 IO23 IO18 IO5 GND
On the ESP32 you can chose which pins you want to use for SPI. The IO0 is between GND and RST.

In one setup I used the following to connect a display:
```
#define TFT_MOSI    0
#define TFT_SCLK    19
#define TFT_CS        23
#define TFT_RST       -1 // 5 // Or set to -1 and connect to Arduino RESET pin
#define TFT_DC         21
```
MISO could be connect to another free pin like IO22, IO18. If you don't use the SD card you may have some other pins left. I used IO13 and IO15 to access the SD-card.

Hope this helps