Ai-Thinker-Open/ESP32-A1S-AudioKit #7: ESP32-A1S-AudioKit V2.2 Keys not responding

Patsjemoe, 2020-08-05:
Hi,
I am running VOIP with  ESP32-A1S-AudioKit, and initialisation and SIP registration is OK .
Unfortunately, the program then does not respond to key inputs.
Using the LyraT v4.3, it works fine, pushing the rec button starts a VOIP call, so it looks like a key-pin defenition that is wrong ?
Also, depending on whether I use audiokit or lyrat, it starts to compile two different files, although it is the same voip file  : 
App "voip_app" version: f8d2671-dirty
App "voip_app" version: f8d2671
Where is the difference coming from  ?
thanks in advance
best regards
Ludo

Patsjemoe, 2020-08-05:
I also wonder if the key interrupt may fail. How to check?

closed 2020-08-06 by Patsjemoe

zhangtemplar, 2022-02-14:
I have the same problem.

This is my definition, however only Key 1 and Key 2 are working.
```c
// According to https://docs.ai-thinker.com/en/esp32-audio-kit
// Only 36 and 13 are working
#define BUTTON_VOLUP_ID           GPIO_NUM_5
#define BUTTON_VOLDOWN_ID         GPIO_NUM_18
#define BUTTON_SET_ID             GPIO_NUM_23
#define BUTTON_PLAY_ID            GPIO_NUM_19
#define BUTTON_MODE_ID            GPIO_NUM_13
#define BUTTON_REC_ID             GPIO_NUM_36
```

PepeTheFroggie, 2023-01-19:
Solution:
    pinMode(KEY3,INPUT_PULLUP);
    pinMode(KEY4,INPUT_PULLUP);
    pinMode(KEY5,INPUT_PULLUP);
    pinMode(KEY6,INPUT_PULLUP);

AbdulSamiKhanOG, 2023-04-16:
@PepeTheFroggie where do i put this