Ai-Thinker-Open/ESP32-A1S-AudioKit #10: Why is this project does not have a common history with espressif/esp-adf ?

msalat, 2020-09-25:
I've copied a few files from ESP32-A1S-AudioKit/components and copy&pasted a few paragraphs components.mk, CMakefile.txt etc. to esp-adf and `idf.py build` works,

but why can't I `git pull https://github.com/espressif/esp-adf` within ESP32-A1S-AudioKit? License incompatibility?

xuhongv, 2020-09-27:
`esp-adf` and `ESP32-A1S-AudioKit` is not the same project.

msalat, 2020-09-30:
Yes, it is not the same project and this is not necesarry, but you could have done `fork` of `esp-adf`, and then we could do

```
git remote add upstream https://github.com/espressif/esp-adf
git fetch upstream # fetch changes from esp-adf
git checkout master
git merge upstream/master
# etc ff
```

Patsjemoe, 2020-10-01:
hi,
you say ...fork of esp-adf...
I am not familiar with this term. where can I find more information on this
?
thanks and regards
ludo

msalat, 2020-10-01:
e. g. here https://github.com/espressif/esp-adf (as on every github project page, this page you are reading now, too) is a button `Fork` where you could create a derivative of an existing github project.

othiman, 2020-10-10:
Hi,

I did this the following way which resulted in the repo https://github.com/othiman/esp-adf.git:

After forking (click fork in the top right corner of https://github.com/espressif/esp-adf) I did the following to achieve this:
```
git clone --recursive https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit.git
git clone --recursive https://github.com/othiman/esp-adf.git
cd ESP32-A1S-AudioKit
git checkout 00e857d63baba188b68e4b61bf5fc1f86ff18fd7
diff -Naru --exclude=.git* --exclude=korvo_du1906 --exclude=lib*.a --exclude=audio_type_def.h --exclude=wav_encoder.c ../esp-adf/ . > ../patch-esp-adf.patch
cd ../esp-adf
git pull origin master
patch -s -p0 < ../patch-esp-adf.patch
git add README-zh.md components/audio_board/ai_thinker_audio_kit_v2_2 components/audio_hal/driver/ac101
git add *
git commit -m "add ai thinker board v2.2 with ac101 audio HAL"
git push origin master
```
After that integrating changes of the upstream repo is as easy as:
```
git remote add upstream https://github.com/espressif/esp-adf.git
git fetch upstream master
git checkout master
git merge upstream/master
```
as msalat mentioned before. The first command of this is only necessary once. Maybe you can also try to get your changes upstream (via a pull request), i.e. your board can be used with the original esp-adf.

Best regards,
othiman

msalat, 2020-10-13:
Fine! Would it be possible to rename your othiman/esp-adf to something 
more clear, perhaps othiman/esp-adf-ac101? There are so many forks of 
esp-adf...

othiman, 2020-10-21:
It did some more changes to my branch. I adapted the board definition to the lyrat_mini_board and it now works fine for the sd card example in `examples/player/element_sdcard_mp3`. Sadly, e.g., the example `examples/player/pipeline_http_mp3` is also working, but stuttering all the time. Will try to open a different issue ticket for this.

msalat, 2020-10-21:
I've tried `examples/advanced_examples/dlna`, stuttiering a lot while playing youtube from browser; a lot better from vlc (os: linux). Also `tried pipeline_living_stream`, worked great for me after search/replace aac -> mp3.

msalat, 2020-10-21:
PS: pipeline_a2dp_sink_and_hfp is running currently on my esp32-a1s-board, but I think the esp32 is a bit oversophisticated for that purpose. 

PS2: `idf.py monitor` showed some error message to pull a patch. Funny that "make" on my linux pc did not mention this.

herostrat, 2021-01-03:
I opened a PR with @othiman branch at esp-adf.
https://github.com/espressif/esp-adf/pull/554

Is this ok?

othiman, 2021-01-04:
I am fine with that. I also already signed the CLA stuff in the PR. Lets see if ESP will integrate it.