Ai-Thinker-Open/ESP32-A1S-AudioKit #11: Use SPI for RC522

Pewibe, 2020-10-19:
I like to attach an RFID RC522 parallel to the onboard SD reader using SPI. 
// SPI GPIOs
#define SD_CS         13    (available at P1.2)
#define SPI_MOSI      15  (available at P1.1)
#define SPI_MISO       2   (no external pin available - wire soldered at SD1.7)
#define SPI_SCK       14    (available at P1.4)
What do you suggest / where to connect for
#define RC522_CS      ???
#define RC522_RST    ???
Any documentation for me?
@schreibfaul1 Do you have the knowledge?

schreibfaul1, 2020-10-19:
I didn't connect a RC522, but a TFT display parallel to the SD card using SPI. The principle is the same.
SPI_MOSI - IO15 - pin MTDO (S5 on)
SPI_MISO - IO2 no pin available, the connection must be established by yourself
SPI_SCK - IO14 - pin MTMS

RC522_CS can be pin IO22
RC522_RST connect with pin RST (EN from A1S)

Pewibe, 2020-10-19:
@schreibfaul1 
IO22 is driving LED4. Have you left this line unchanged for your TFT or did you cut R14 that there is no load on IO22?
Actually our problem is that SD and RC522 are working well at SPI (MOSI, MISO, SCK parallel - CS by 2 different IOs) for their own. As soon as the other device is initialized by the software, a problem occurs.

schreibfaul1, 2020-10-19:
I haven't changed anything on the LEDs. LED4 lights up when CS is active (IO22 is low). This does not affect the function.

Pewibe, 2020-10-28:
In the meantime both devices (SD and RFID) work together at SPI. I think the originally problem was a 2nd (erroneous) initialization of the SPI.

closed 2020-10-28 by Pewibe