Ai-Thinker-Open/ESP32-A1S-AudioKit #12: examples/get-started/pipeline_a2dp_sink_and_hfp crashes on ESP32-AiThinker-audio V2.2 due to RTC error

sellerie98, 2020-11-14:
I set up the toolchain according to the guide and tried to compile the non-modified pipeline_a2dp_sink_and_hfp application from the examples. I copied the `examples` folder from the repo root to `esp-idf/ai-examples` and tried to compile it using:
`make menuconfig` and ` make PYTHON=python2 flash monitor`.

I get the following error:
```
E (931) BT_BTM: btm_sco_process_num_bufs, 4
I (1001) BLUETOOTH_EXAMPLE: [ 2 ] Start codec chip
I (1001) A1S: i2c port configured!!!!
W (2001) AC101: reset succeed
I (2011) AC101: init done
I (2011) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2011) AUDIO_HAL: Codec mode is 3, Ctrl:1
I (2121) BLUETOOTH_EXAMPLE: [ 3 ] Create audio pipeline for playback
I (2121) BLUETOOTH_EXAMPLE: [3.1] Create i2s stream to write data to codec chip and read data from codec chip
I (2131) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (2131) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (2141) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
I (2151) A1S: i2s port configured!!!!
W (2161) I2S: I2S driver already installed
I (2161) A1S: i2s port configured!!!!
I (2171) BLUETOOTH_EXAMPLE: [3.2] Get Bluetooth stream
I (2171) BLUETOOTH_EXAMPLE: [3.2] Register all elements to audio pipeline
I (2181) BLUETOOTH_EXAMPLE: [3.3] Link it together [Bluetooth]-->bt_stream_reader-->i2s_stream_writer-->[codec_chip]
I (2191) AUDIO_PIPELINE: link el->rb, el:0x3ffdca1c, tag:bt, rb:0x3ffdcda8
I (2201) AUDIO_PIPELINE: link el->rb, el:0x3ffdc3d4, tag:i2s, rb:0x3ffdeee4
I (2211) BLUETOOTH_EXAMPLE: [ 4 ] Initialize peripherals
I (2211) BLUETOOTH_EXAMPLE: [4.1] Initialize Touch peripheral
I (2221) BLUETOOTH_EXAMPLE: [4.2] Create Bluetooth peripheral
I (2221) BLUETOOTH_EXAMPLE: [4.2] Start all peripherals
E (2231) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:813 (touch_pad_config):Touch_Pad Num Err
E (2241) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:813 (touch_pad_config):Touch_Pad Num Err
E (2261) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:813 (touch_pad_config):Touch_Pad Num Err
W (2271) PERIPH_TOUCH: _touch_init
I (2271) BLUETOOTH_EXAMPLE: [ 5 ] Set up  event listener
I (2281) BLUETOOTH_EXAMPLE: [5.1] Listening event from all elements of pipeline
I (2291) BLUETOOTH_EXAMPLE: [5.2] Listening event from peripherals
I (2291) BLUETOOTH_EXAMPLE: [ 6 ] Start audio_pipeline
I (2301) AUDIO_ELEMENT: [bt-0x3ffdca1c] Element task created
I (2311) AUDIO_ELEMENT: [i2s-0x3ffdc048] Element task created
I (2311) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:358, MEM Total:111796 Bytes

I (2321) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (2331) I2S_STREAM: AUDIO_STREAM_WRITER
I (2341) AUDIO_PIPELINE: Pipeline started
I (2341) AUDIO_ELEMENT: [i2s-0x3ffdc3d4] Element task created
I (2341) AUDIO_ELEMENT: [raw-0x3ffdc6f4] Element task created
I (2351) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:358, MEM Total:105796 Bytes

I (2361) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (2361) I2S_STREAM: AUDIO_STREAM_READER,Rate:44100,ch:2
I (2371) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
I (2391) AUDIO_PIPELINE: Pipeline started
I (2391) BLUETOOTH_EXAMPLE: [ 7 ] Listen for all pipeline events
E (2421) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (2421) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (2431) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (2571) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (2571) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (2581) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (2721) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (2721) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (2731) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (2871) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (2871) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (2881) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (3021) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (3021) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (3031) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (3171) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (3171) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
[... 9 lines cut ...]
E (3631) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (3771) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (3771) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (3781) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (3921) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (3921) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (3931) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (4071) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (4071) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (4081) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (4221) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (4221) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (4231) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (4371) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (4371) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (4381) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
[... cut ...]
E (4671) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (4681) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (4821) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
E (4821) RTC_MODULE: /home/sellerie/git/ESP32-A1S-AudioKit/esp-idf/components/driver/rtc_module.c:936 (touch_pad_read_filtered):Touch_Pad Num Err
```

c-logic, 2020-12-11:
deactivate touch-components for test.

sellerie98, 2021-01-09:
> deactivate touch-components for test in source-file.

In what source file?

c-logic, 2021-01-09:
as example examples/get-started/pipeline_a2dp_sink_and_hfp/main/a2dp_sink_and_hfp_example.c

i have the esp32 audio kit v2.2 without touch-controls. its a dirty patch.