Ai-Thinker-Open/ESP32-A1S-AudioKit #18: Line Input / Mic Output Max Levels

benjmarshall, 2021-02-19:
Hi, I'm trying to work out the maximum rating for the line level input, and maximum output level for the headphone output. I can't seem to find this information in the docs available. Could any provide me with any details?

Ideally I'd like to use the A1S audio kit board in a guitar amp effects loop which can operate at either -10dBv or +4dBv, could any provide any information on if these signal levels will work with the audio kit?

Thanks!

cgreening, 2021-02-24:
I'd like to know this as well, I'm about to do some testing and don't want to blow it up by driving it too hard.

Looking at the AC101 datasheet I can see in the specs:

> One differential or single-ended line-in input

In the AC101 manual it says:

> LINEINL/R provide one-channel mono differential input or stereo single-ended input that can be mixed into the ADC record mixer or the stereo output mixer. The inputs are suited to receiving line level signals such as external audio equipment or baseband module.

So - not a lot of help really! Depends on how you interpret "line level signals". https://en.wikipedia.org/wiki/Line_level

Feels like you'd be safe starting off with -10dBv and if that's way too quiet go up to +4dBv.

benjmarshall, 2021-02-24:
Hey @cgreening, I've started doing some testing on this. I'm using the yummyDSP library (which in turn uses the ESP32 Arduino lib) as a base. See [here](https://github.com/benjmarshall/ai_thinker_esp32_audio_kit/tree/main/line_in_testing) for a start on a program to figure out how the ADC responds.

So far I've driven the line in from a usc-dac from a phone which seems to be easily within the acceptable input range. I've used the linked program to determine that at "max" volume my dac drives the ADC to approximately 50% of full output level when i play test tones.

I will try and get out the multi meter tomorrow and see if I can get an idea of how the voltage on the line in port correlates to ADC values. I think i will need a much beefier input to start to clip the input. I've got a few other DAC's that should be able to drive a better voltage out to test with before I risk the guitar amp! I'll post up here if I can supply any more useful information.

Side note: I think the headphone output has been wired up with L/R channels reversed? I have had to route the L/R channels to the "wrong" DAC channels to get L/R to appear in my headphones the way that the tone generator says they should be. A quick look at the schematic for the board does seem to show the pins on the line in and headphones sockets are switched for some reason!

cgreening, 2021-02-25:
Thanks @benjmarshall that is really useful - I'd been reading all sorts of horrible things about having to write to registers to enable the line-in input! I'll definitely be making use of your sample code. I should get some time this weekend so I'll plug in the signal generator and see what comes it does.

benjmarshall, 2021-02-25:
I've taken some initial measurements this evening, raw data below:

0.19f recorded by ADC - 288mV AC
0.63f recorded by ADC - 835mV AC
0.85f recorded by ADC - 1090mV AC
0.99.f(max) recorded by ADC - 1400mV AC (Clipping heard)

I used the first 3 values to plot a curve fit to estimate what the full scale limit roughly is (4th sample is beyond the full scale ADC limit). From the graph i create it looks to be about 1.25V.

The ADC, boosts, mix path etc were configured as per the code I linked before. From my understanding of the data sheet this should basically be the "default" mode of the ADC. I believe there is scope to apply different levels of gain on the line in signal before it hits the ADC which should allow you to change this voltage range to some extent.

So -10dBv will be fine for the ADC but will use < 1/3rd of the default input range so would need some playing with the gain settings. I reckon it's likely that +4dBv won't be an issue for the electronics (1.4V) seems fine, but will definitely clip when using the default setting, so will also need some playing with gain settings (i think the default gain can be reduced, or maybe it's an attenuation?).

More experimentation to be done!

benjmarshall, 2021-02-25:
I think register 0x41h - ADC Volume Control will allow a gain or attenuation at the input to the ADC, default is 0dB and seems to be configurable across a wide range in 0.75dB increments, so getting to a suitable level for either -10dBv or +4dBv should be fine.

cgreening, 2021-02-26:
This is brilliant! You’ve saved me an awful lot of time.

benjmarshall, 2021-02-26:
All good @cgreening! I'm hoping to play with this some more and test the output side too.

If you discover anything or get somewhere with your own testing please do post up here so we track all the info!

Also, if your playing with the board in a similar vein to what I'm trying to achieve then please ping me a link to what your working on (assuming your sharing the code) as any inspiration is always super helpful!

benjmarshall, 2021-02-26:
I've made some further updates to my code over at https://github.com/benjmarshall/ai_thinker_esp32_audio_kit/tree/main/line_in_testing which should help you @cgreening. I've added the ability to adjust the ADC gain up and down on the fly. The code also prints out a rolling maximum value being seen at the ADC to help make adjustments. This is working well for me playing with feeding music in from a USB DAC, I can up the gain to compensate for the slightly low signal levels and watch the peak values increasing. I've pushed it to the point where it starts to clip pretty badly and don't seem to have inflicted any damage yet!