Ai-Thinker-Open/ESP32-A1S-AudioKit #19: >

jalakamkiranadspl, 2021-03-24:


jalakamkiranadspl, 2021-03-31:
To who ever is facing this issue.Since no one from the AI thinker had responded I had to get the external repository from github. 
@chandrabezzo has created a repo for porting the esp adf for AI thinker board and ported some of the examples. I forked it from him and have made some deprecation changes. You can find the repo linked below.
[](https://github.com/jalakamkiranadspl/Ai-thinker-audio-kit)

closed 2021-03-31 by jalakamkiranadspl

chandrabezzo, 2021-04-01:
Hi @jalakamkiranadspl thank you for your reporting. To appreciate me can you fork out of this repo and make changes? Not clone then publish in your repo.

jalakamkiranadspl, 2021-04-01:
Hey Chandra bezzo...I will definitely do that.

jalakamkiranadspl, 2021-04-01:
Also can you check my other email I have written to you regarding an issue
with Speech recognition ASR. Please help me out with that.
Thanks.
I am quite new to github. So I did what I thought was good (mentioning you
in comments) sorry about not forking your repo at first itself.

jalakamkiranadspl, 2021-04-01:
@chandrabezzo  I have forked and done the same. Please check it out. :)