Ai-Thinker-Open/ESP32-A1S-AudioKit #26: Codec initialization fails

mihovilkolaric, 2021-07-07:
I bought a "ESP32 Audio Kit v2.2 2957" and installed the toolchain as described in this readly, and the ADF (+IDF) from this repo.
I can compile and flash all examples I tried so far (here, e.g. ESP32-A1S-AudioKit/examples/get-started/play_mp3), but when running them, I find following errors on the serial console:
```
E (1026) AC101: reset failed!
E (1026) AUDIO_HAL: codec init failed!
E (1026) AUDIO_BOARD: /home/test/esp/ESP32-A1S-AudioKit/components/audio_board/ai_thinker_audio_kit_v2_2/board.c:55 (audio_board_codec_init): Got NULL Pointer
E (1036) AUDIO_HAL: audio_hal handle is null
```
and do not hear anything in the earphones.

I verified that in `make menuconfig` the board `ESP32-AiThinker-audio V2.2` is selected.

What can I do to fix this issue?

PratyushGehlot, 2021-07-11:
Hi  @mihovilkolaric ,
I also bought last week  `ESP32 Audio Kit v2.2 2957 `  and also facing the same issue. 
I used Arduino sketch from here - https://github.com/schreibfaul1/AC101/blob/master/examples/ESP32AudioKit/ESP32AudioKit.ino

AC101 init failing - 
> Serial.printf("Connect to AC101 codec... ");
while (not ac.begin(IIC_DATA, IIC_CLK))   
	{
		Serial.printf("Failed!\n");
		delay(1000);
	}

Serial Output : `Connect to AC101 codec... Failed!`

Setup : 
[image]

xpeace, 2021-07-13:
Recent AI Thinker boards V2.2 2957 are using A1S designs where the AC101 was replaced by the ES8388 eventually these are the wrong boards for this A1S version. It could be possible that all recent boards are defective if the outputs of the A1S-ES8388 are not compatible to the 2.2 Audio Kit board... I did not receive an official statement regarding this mess yet... please ask the AI Thinker support. However I have created an Arduino file which autodetecs your HW wether you're having an AC101 or the ES8388 based design... AI Thinker confirmed they created these A1S ES8388 based chips... they gave no answer why tho. You see the chip when removing the EM shield - or upload the arduino file...
 https://www.mikrocontroller.net/topic/474383?goto=6754783#6752864

maqsudbd, 2021-07-15:
Hi @PratyushGehlot  am facing the same issue.

Did you find any solution?

Hi @xpeace

Did you mean that the board is faulty? Or we will have to use driver for ES8388?

mihovilkolaric, 2021-07-15:
@maqsudbd : when using the driver for ES8388 (tried this by changing the "Audio Board" to LyraT 4.3 in make menuconfig), the codec can be initialized, but on the headphones-jack of the audio-board you hear only one channel, the volume is very low (hardly hearable), and can not be controlled by software.
So, hard to tell whether the board is faulty (resp.: pinning of board and chip are incompatible), or whether some adjustments in the code/firmware/GPIO-config are enough to make it work. It is impossible to say, without having a datasheet of the used A1S (with ES8388).

maqsudbd, 2021-07-15:
@mihovilkolaric Thanks for update.

I was planning to use Arduino, made a code for AC101 and I was able to made it work for a friend. Now I am stuck and hope someone will find a solution.

I think datasheet of the used A1S (with ES8388) will help to find the root cause. I hope the board is not faulty by design.

xpeace, 2021-07-15:
Eventually GPIO 21 needs to be activated to enable the amp...
Still trying to get in contact with AI Thinker... had not much success lately...

xpeace, 2021-07-16:
Additionally I`ve found this pdf reflecting changes in the A1S board...

https://www.docdroid.net/L6LX9SK/esp32-a1s-esv23zh-cnen-pdf
but I think they left out the correct values when describing the I²C / I²S values... for the esp-adf 

/* I2C gpios */
#define IIC_CLK                     23
#define IIC_DATA                   18

It does also mention a V2.3 Audio Kit - while AI Thinker put them on a V2.2 pcb...

xpeace, 2021-07-17:
Most recent V2.3 pdf... https://docs.ai-thinker.com/_media/esp32-a1s_v2.3_specification.pdf

maqsudbd, 2021-07-17:
@xpeace Thanks for sharing the pdf.

It seems the board is not faulty rather a new one with new design. I2C are now on different pins (23,18 now, before 32,33). Also the I2S pins are different.

tttin46, 2021-09-25:
I found this topic in Chinese: 
https://blog.csdn.net/Boantong_/article/details/117479162?ops_request_misc=%257B%2522request%255Fid%2522%253A%2522163258324316780269812827%2522%252C%2522scm%2522%253A%252220140713.130102334.pc%255Fall.%2522%257D
Hope this help

xpeace, 2021-09-25:
Apart from headphone recognition I got all the functions I need working. 
To get max volume from the mic of the ES8388 unit it requires to enable AGC.

tttin46, 2021-09-26:
New modified ESP-ADF git: https://github.com/SeaHi-Mo/esp-adf

rxd1999, 2021-10-06:
> Apart from headphone recognition I got all the functions I need working. To get max volume from the mic of the ES8388 unit it requires to enable AGC.

Hi
I meet the same problem these days.
Could you tell me how to enable "AGC"?
Thanks !

tttin46, 2021-10-06:
> > Apart from headphone recognition I got all the functions I need working. To get max volume from the mic of the ES8388 unit it requires to enable AGC.
> 
> Hi I meet the same problem these days. Could you tell me how to enable "AGC"? Thanks !

AMP is controlled by IO21 but its use for Speaker not for headphone jack.
[image]

xpeace, 2021-10-06:
I think the question is not about the amp, to increase the mic input sensitivity AGC (Automatic Gain Control) needs to be enabled. Unfortunately it is not possible to increase the pga (programmable gain amplifier) unit without enabling AGC and adjusting the corresponding values. 

Here is the settings section I`m using to test the mic input, enable the deemphasis to reduce the quantization noise.
And try to reduce any heavy calculation during capture :)... Imho the biggest "feature" is you got the cpu`s impacting the AD signal path. Boot buttons and Key6 is also impacting the capture - anyone knows what`s up here ?

    // PGA ALC/AGC
	es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL10, 0xDA);  // Reg 0x12 = 0xE2 (ALC enable, PGA Max. Gain=23.5dB, Min. Gain=0dB)
	es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL11, 0x91);  // Reg 0x13 = 0xC0 (ALC Target=-4.5dB, ALC Hold time = 0 mS)
    es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL12, 0x12);  // Reg 0x14 = 0x12 (Decay time =820uS , Attack time = 416 uS)
	es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL13, 0x06);  // Reg 0x15 = 0x06 (ALC mode) 96 Samples
	es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL14, 0x00);  // Reg 0x16 = 0xC3 (noise gate = -40.5dB, NGG = 0x01(mute ADC))

GillesLACAUD, 2021-10-20:
If they change the codec and change also the pinout between the different productions....

GillesLACAUD, 2021-10-20:
I can find any schematic with the I2C 33/32 and the ES8388 !!!

GillesLACAUD, 2021-10-21:
Hi Xuhongv.
Thank you for your quick answer. It will be very usefull.
Can you also upload the last schematic or data sheet in this post or on your web site, and give us the link.
Is it the final version of the ESP32 Audio Kit or you plan to change something else in the futur ?
I want to buy a lot of your cards and i must be sure to have always the same boards.

GillesLACAUD, 2021-10-21:
I am sorry but i am a little bit lost...
Where is the github with your information.

When i search i find this: 

For the ADF github
[https://github.com/espressif/esp-adf/blob/master/components/audio_board/lyrat_v4_3/board_pins_config.c](url)
And For the Audio kit github
[https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit/blob/master/components/audio_board/lyrat_v4_3/board_pins_config.c](url)

[image]

xuhongv, 2021-10-21:
[esp32-a1s_v2.3-20210508(1).pdf](https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit/files/7387472/esp32-a1s_v2.3-20210508.1.pdf)

mdhosale, 2021-11-03:
Hi all-
(sorry this is so complicated....)
so I made some progress on the AiThinker with the ES8388, but I am having an issue where the sound is basically working, but I am hearing a low hiss in the right channel that eventually gets louder and louder - I think the clock timing is off, but I am not sure where to dig into debugging that. i am about 90-95% there. Maybe someone has some tips on what may be wrong? I have had good luck using the AC101 BTW, this ES8388 has thrown me for a loop. Here’s what I found:

Note: I am using Arduino IDE with ESP32 boards 1.6 (2.0 is not working)
I am using code base don the FAUST IDE: https://faustdoc.grame.fr/tutorials/esp32/
this is for a project that takes audio in, processes it via a bank of filters, then outputs it out the headphone jack.

The pinout is different for the Aithinker board so I made the following changes:

in the faust outputted .cpp file I made the pinout changes suggested above :
```

#define LYRA_T true

then where is reads:
#elif LYRA_T
            pin_config = {
                .bck_io_num = 5,
                .ws_io_num = 25,
                .data_out_num = 26,
                .data_in_num = 35
            };
```
            
change to:

```
#elif LYRA_T
            pin_config = {
                .bck_io_num = 27,
                .ws_io_num = 25,
                .data_out_num = 26,
                .data_in_num = 35
            };  
```

then in es388.h I changed, more pinout stuff:

```
#define I2C_MASTER_NUM I2C_NUM_1 /*!< I2C port number for master dev */

#define I2C_MASTER_SCL_IO GPIO_NUM_23//these may be wrong 
#define I2C_MASTER_SDA_IO GPIO_NUM_18

#define I2C_MASTER_FREQ_HZ 100000
#define I2C_MASTER_TX_BUF_DISABLE 0
#define I2C_MASTER_RX_BUF_DISABLE 0
 

#define WRITE_BIT I2C_MASTER_WRITE              /*!< I2C master write */
#define ACK_CHECK_EN 0x1
```

to

```
//from the example. nessecary info for I2C. common protocol defines
#define WRITE_BIT                          I2C_MASTER_WRITE /*!< I2C master write */
#define READ_BIT                           I2C_MASTER_READ  /*!< I2C master read */
#define ACK_CHECK_EN                       0x1              /*!< I2C master will check ack from slave*/
#define ACK_CHECK_DIS                      0x0              /*!< I2C master will not check ack from slave */
#define ACK_VAL                            (i2c_ack_type_t)0x0              /*!< I2C ack value */
#define NACK_VAL                           (i2c_ack_type_t)0x1              /*!< I2C nack value */

#define I2C_TX_BUF_DISABLE    0                /*!< I2C master do not need buffer */
#define I2C_RX_BUF_DISABLE    0                /*!< I2C master do not need buffer */

#define PA_ENABLE_GPIO          GPIO_NUM_21 //a relevent GPIO pin for the LYRAT 4.2, for turning the PA power on

[... 102 lines cut ...]
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL3/*25, 0x19*/,     0x00);  // 32 disable volume control soft ramp, normal - check DAC mute if you hear inputs
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL6/*28, 0x1c*/,     0x08); //phase invert off
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL7/*29, 0x1d*/,     0x00); //00 normal stereo, streength default
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL4/*26, 0x1a*/,     0x00); //left vol 0dB
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL5/*27, 0x1b*/,     0x00); // right vol 0dB

  
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL16/*38, 0x26*/,     0x00); //select for output mix - default
//res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL23/*45,0x2d*/, 0x00);   //vroi=0
  //02
  res |= es_write_reg(ES8388_ADDR, ES8388_CHIPPOWER/*2, 0x02*/, 0x00); // normal all and power up all

  return res;
}
```

In Arduino I get this running by declaring the following:
```

 es.es8388_startTest();

  delay(3000);
  // start faust dsp
  faustcode.start();
```

mdhosale, 2021-11-04:
Hi all- 
I think I have a better way to explain this.
I made some progress on the AiThinker with the ES388, but I am having an issue where the sound is working, but I am hearing a low hiss in the right channel that eventually gets louder and louder, I am not sure where to beg into debug that. i am about 90-95% there. Maybe someone has some tips on what may be wrong. Here’s what I found:

Note: I am using Arduino IDE with ESP32 boards 1.6 (2.0 is not working)
this is for a project that takes audio in, processes it via a bank of filters, then outputs it out the headphone jack.

I am working based on the faust esp32 example found here: https://faustdoc.grame.fr/tutorials/esp32/

The pinouts on the AiThinker with an E8388 need to be changed as per the image provided above. So, I made the following changes,

in the Faust generated .cpp file :
`#define LYRA_T true`

then where is reads:
```
#elif LYRA_T
            pin_config = {
                .bck_io_num = 5,
                .ws_io_num = 25,
                .data_out_num = 26,
                .data_in_num = 35
            };
            
```
change to:

```
#elif LYRA_T
            pin_config = {
                .bck_io_num = 27,
                .ws_io_num = 25,
                .data_out_num = 26,
                .data_in_num = 35
            };  
```

then in es388.h change, more pinout stuff:

```
#define I2C_MASTER_NUM I2C_NUM_1 /*!< I2C port number for master dev */

#define I2C_MASTER_SCL_IO GPIO_NUM_23//these are wrong 
#define I2C_MASTER_SDA_IO GPIO_NUM_18

#define I2C_MASTER_FREQ_HZ 100000 
#define I2C_MASTER_TX_BUF_DISABLE 0
#define I2C_MASTER_RX_BUF_DISABLE 0
 

#define WRITE_BIT I2C_MASTER_WRITE              /*!< I2C master write */
#define ACK_CHECK_EN 0x1
```

to 

```
//from the example. nessecary info for I2C. common protocol defines
#define WRITE_BIT                          I2C_MASTER_WRITE /*!< I2C master write */
#define READ_BIT                           I2C_MASTER_READ  /*!< I2C master read */
#define ACK_CHECK_EN                       0x1              /!< I2C master will check ack from slave/
#define ACK_CHECK_DIS                      0x0              /*!< I2C master will not check ack from slave */
#define ACK_VAL                            (i2c_ack_type_t)0x0              /*!< I2C ack value */
#define NACK_VAL                           (i2c_ack_type_t)0x1              /*!< I2C nack value */

#define I2C_TX_BUF_DISABLE    0                /*!< I2C master do not need buffer */
#define I2C_RX_BUF_DISABLE    0                /*!< I2C master do not need buffer */

#define PA_ENABLE_GPIO          GPIO_NUM_21 //a relevent GPIO pin for the LYRAT 4.2, for turning the PA power on

[... 99 lines cut ...]
  
  res |= es_write_reg(ES8388_ADDR, ES8388_CHIPPOWER/*2, 0x02*/, 0x00); // normal all and power up all

  //---------------------
  //--not used? res |= es_write_reg(ES8388_ADDR, ES8388_CHIPLOPOW1/*5,0x05*/,0x00);
  //--not used? res |= es_write_reg(ES8388_ADDR, ES8388_CHIPLOPOW2/*6, 0x06*/,0x00);
  //--not used? res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL3/*25, 0x19*/,     0x00);  // 32 disable volume control soft ramp, normal - check DAC mute if you hear inputs
  //--not used? res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL6/*28, 0x1c*/,     0x08); //phase invert off
  //--not used? res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL7/*29, 0x1d*/,     0x00); //00 normal stereo, streength default
  //--not used? res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL23/*45,0x2d*/, 0x00);   //VREF to analog output resistance
  //---------------------

  return res;
}
```

In Arduino I get this running by declaring the following:

```
 es.es8388_startTest();

  delay(3000);
  // start faust dsp
  faustcode.start();
```

GillesLACAUD, 2021-11-04:
@xuhongv No idea about the 2957 boards ? It is very strange...

GillesLACAUD, 2021-11-05:
@mdhosale
Could you please give us the serial number of yours ESP32 Audio Kit ? Thank you

mdhosale, 2021-11-05:
> @mdhosale Could you please give us the serial number of yours ESP32 Audio Kit ? Thank you

it is also the 2974 board. thank you!

GillesLACAUD, 2021-11-05:
So we have three serial Number
- 2957
- 2974
- 3378

GillesLACAUD, 2021-11-08:
@xuhongv 
Could you confirm this please ?
- xxxx  Codec AC101 with I2C pin 32/33 
- 2957 Codec ES8388 with I2C pin 18/23
- 2975 Codec ES8388 with I2C pin 32/33
- 3378 Codec ES8388 with I2C pin 32/33

Thank you

So the 2957 and 2975/3378 are not compatible.
Me I use the pin 18/23 as an UART to control a Nextion screen. So when I receive a **2957 board I can not use it**.
You have stop the production of the 2957 (pin 18/23) but the provider have always these boards in stock.
So when I receive the boards I just hope to have the good serial numbers.... :-(

tttin46, 2021-11-09:
Finally I found this solution for low output volume at earphone jack. Just modify ADF audio hall 8366 driver (esp-adf\components\audio_hal\driver\es8388\es8388.c)
Find "es8388_set_voice_volume" function and modify it like this:

esp_err_t es8388_set_voice_volume(int volume)
{
    esp_err_t res = ESP_OK;
    if (volume < 0)
        volume = 0;
    else if (volume > 100)
        volume = 100;
    volume /= 3;
	res = es_write_reg(ES8388_ADDR, ES8388_DACCONTROL4, volume);
	res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL5, volume);
    res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL24, volume);
    res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL25, volume);
    res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL26, volume);
    res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL27, volume);
    return res;
}

vanbwodonk, 2021-11-09:
> Thank you
> 
> @xuhongv Could you confirm this please ?
> 
> * xxxx  Codec AC101 with I2C pin 32/33
> * 2957 Codec ES8388 with I2C pin 18/23
> * 2974 Codec ES8388 with I2C pin 32/33
> * 3378 Codec ES8388 with I2C pin 32/33
> 
> Thank you
> 
> So the 2957 and 2975/3378 are not compatible. Me I use the pin 18/23 as an UART to control a Nextion screen. So when I receive a **2957 board I can not use it**. You have stop the production of the 2957 (pin 18/23) but the provider have always these boards in stock. So when I receive the boards I just hope to have the good serial numbers.... :-(

 @GillesLACAUD, i have 2 board 2974 with Codec ES8388,  with i2c pin 18/23 and 32/33. I'm so confused.

GillesLACAUD, 2021-11-09:
@vanbwodonk

I have receive 5 boards yesterday.
3 Boards with 2974 serial number
2 Boards with 2957 serial number
And all the boards are OK....I'm so confused too. 

Maybe we should check the two serial number
- Serial Number for the audio board (under the switch)
- serial Number for the A1S module (under the antenna)

Can you please send some picture of the two boards.
I am in touch with the AI support and they ask me to send them some picture of my fail boards...but I have return them yesterday.

Best

vanbwodonk, 2021-11-09:
> res = es_write_reg(ES8388_ADDR, ES8388_DACCONTROL4, volume);
> res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL5, volume);
@tttin46 Are you sure? DACCONTROL4 & DACCONTROL5 is attenuation. Keep it 0x00.

vanbwodonk, 2021-11-09:
@GillesLACAUD

- audio board SN (under the switch) -> V2.2 2762(32/33) and V2.2 2957(18/23). 
- A1S board is same (under the antenna) SN -> 2974.

GillesLACAUD, 2021-11-09:
Thank you vanbwodonk
According to an email from AI thinker
"
_Dear Gilles,

Thank you for your email and sorry for my late reply.
I am wanna help you settle the problem but not able to contact you via Facebook.
Pls find the attached file, maybe it is helpful for you

About what you mentioned in the last email, could you pls take a photo of the backside for us?
Our engineer want to confirm something_"

Could you please send us a photo of the backside for the V2.2 2957 (18/23)

Thx

tttin46, 2021-11-09:
> > res = es_write_reg(ES8388_ADDR, ES8388_DACCONTROL4, volume);
> > res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL5, volume);
> 
> @tttin46 Are you sure? DACCONTROL4 & DACCONTROL4 is attenuation. Keep it 0x00. I make spreadsheet for es8388 control register, hope this help others. https://docs.google.com/spreadsheets/d/1_UdXtrtpj_dzi36CSwv5tBphtGsBrY2Npdi79h1dYUk/edit?usp=sharing

DACCONTROL4 & DACCONTROL4 are digital volume control registers for both output 1&2 while DACCONTROL24-27 are separately control register for each output.

mdhosale, 2021-11-09:
Hi Everyone - 
I have everything working, that is ADC -> FAUST Code -> DAC in Arduino
- except -
1) I can hear a low level of the input coming through the DAC from the ADC- I cannot locate any register to control that (besides 0x27  and 0x2a - but this is not the issue)
2) the microphone on the board is turning on along with the line in. So far I can’t find where in the registers this could be happening.  Is the board is wired with the line in and microphone on the same input for some reason? I found the schematic for this board above, but the line in and mic wiring details are not 100% clear to me-

The registers I am using are below- maybe someone can see something? - thanks in advance for your help!!

 ```
/* Chip Control and Power Management */
  res |= es_write_reg(ES8388_ADDR, ES8388_MASTERMODE/*8, 0x08*/, 0x00); //84 master mode no dac heardCODEC IN I2S SLAVE MODE MCLK/4 
  res |= es_write_reg(ES8388_ADDR, ES8388_CHIPPOWER/*2, 0x02*/, 0xFF); //power down to set registers
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL21/*43, 0x2b*/, 0x80); //80 or 00?set internal ADC and DAC use the same LRCK clock, ADC LRCK as internal LRCK
 
  res |= es_write_reg(ES8388_ADDR, ES8388_CONTROL1/*0, 0x00*/, 0x07);  //05, 12, 17 all work?12 Enfr=0,Play&Record Mode,(0x17-both of mic&paly)
  res |= es_write_reg(ES8388_ADDR, ES8388_CONTROL2/*1, 0x01*/, 0x00); //40,00? normal power
  res |= es_write_reg(ES8388_ADDR, ES8388_ADCPOWER/*3, 0x03*/, 0x08);//0x08?microphone bias power down (high impedance output, default)
  res |= es_write_reg(ES8388_ADDR, ES8388_DACPOWER/*4, 0x04*/, 0x3C);//DAC Power Management
   
  //ADC
  res |= es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL2/*10, 0x0a*/, 0x50); //line 2 0x50, *0x00 (single ended input), 0xF0 (differential and Pseudo Differential), check this - for noise?? ADC_INPUT_DIFFERENCE 0x00 LINSEL & RINSEL, LIN1/RIN1 as ADC Input; DSSEL,use one DS Reg11; DSR, LINPUT1-RINPUT1
  res |= es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL3/*11, 0x0b*/, 0x02);//bypassed? 0x02 (Select LIN1and RIN1 as differential input pairs),0x82 is LIN2 and RIN2
 
  res |= es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL1/*9,0x09*/, 0x00);// PGA gain - don't change - no boost desired
  res |= es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL4/*12, 0x0c*/, 0x00);// 00 *0x0c = 16bits, 0x40 in example seems wrong, note says "Reg 0x0C = 0x00 (I2S – 24bits, Ldata = LADC, Rdata = RADC)" Left/Right data, Left/Right justified mode, Bits length, I2S format
  res |= es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL5/*13, 0x0d*/, 0x02);// clock ratio 256

  //added
  res |= es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL6/*14, 0x0e*/, 0x00);//default 0x30- polarity and HPF, 00 disable HPF
  res |= es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL7/*15, 0x0f*/, 0x30);//default 0x20- gain parameters, muting -- disabled digital volume control soft ramp
  
  res |= es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL8/*16, 0x10*/, 0x06);// L Digital volume control attenuates the signal in 0.5 dB incremental from 0 to –96 dB.
  res |= es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL9/*17, 0x11*/, 0x06);// R Digital volume control attenuates the signal in 0.5 dB incremental from 0 to –96 dB.

//  //ALC
  res |= es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL10/*18, 0x12*/, 0x00);   // 0x00==disable, F8 full range ALC on- Reg 0x12 = 0xe2(alc enable, PGA Max. Gain=23.5dB, Min. Gain=0dB) 
  res |= es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL11/*19, 0x13*/, 0xF0);   // a0 - Reg 0x13 = 0xc0 (ALC Target=-4.5dB, ALC Hold time =0 mS)
  res |= es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL12/*20, 0x14*/, 0x12);   // Reg 0x14 = 0x12(Decay time =820uS , Attack time = 416 uS)
  res |= es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL13/*21, 0x15*/, 0x06);   // Reg 0x15 = 0x06(ALC mode) 
  res |= es_write_reg(ES8388_ADDR, ES8388_ADCCONTROL14/*22, 0x16*/, 0xc3);   // Reg 0x16 = 0xc3(nose gate = -40.5dB, NGG = 0x01(mute ADC))
  
  //DAC
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL1/*23, 0x17*/,      0x00);// 00 - 17 - 1a 0x18:16bit iis , 0x00:24
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL2/*24, 0x18*/,      0x02);  //DACFsMode,SINGLE SPEED; DACFsRatio,256

  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL3/*25, 0x19*/,      0x32);  // DAC mute == 0x04 
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL4/*26, 0x1a*/,      0x00); //L Digital volume control attenuates the signal in 0.5 dB incremental from 0 to –96 dB.
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL5/*27, 0x1b*/,      0x00); //R Digital volume control attenuates the signal in 0.5 dB incremental from 0 to –96 dB.
  
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL16/*38, 0x26*/,     0x09); //1B or 1C - select for output mix - default
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL17/*39, 0x27*/,     0xBF);//*-B8 L DAC mixer enable, set to 0dB
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL18/*40  0x28*/,     0x38);//??Phase inversion controls
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL19/*41  0x29*/,     0x38);//??stereo - mono controls
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL20/*42, 0x2a*/,     0xBF);//-B8 R mixer enable, set to 0dB
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL24/*46, 0x2e*/,     0x1E);// Lout1 0dB
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL25/*47, 0x2f*/,     0x1E);// Rout1 0dB
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL26/*48, 0x30*/,     0x1E);// Lout2 0dB
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL27/*49, 0x31*/,     0x1E);// Lout2 0dB

  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL6/*28, 0x1c*/,      0x08); //phase invert off
  res |= es_write_reg(ES8388_ADDR, ES8388_DACCONTROL7/*29, 0x1d*/,      0x00); //00 normal stereo, strength default, 40 mute R
  
  res |= es_write_reg(ES8388_ADDR, ES8388_CHIPPOWER/*2, 0x02*/, 0x00); // normal all and power up all
```

vanbwodonk, 2021-11-10:
@GillesLACAUD 
[image]
[image]

GillesLACAUD, 2021-11-12:
@vanbwodonk 
Thank you very much

Receive 3 boards today
For all boards ESP32 Audio Kit V2.2 3378 and for the ESP32 A1S (under the antenna) 2974

1 Board with ES8388 map at I2C 32/33 
2 Boards with ES8388 mat at I2C 18/23

@xuhongv
**IT IS ABSOLUTLY INCREDIBLE...WE BUY A PRODUCT AND WE HAVE DIFFERENT MAPPING FOR THE IO...
XUHINGV PLEASE EXPLAIN THIS AND FIND A SOLUTION**

GillesLACAUD, 2021-11-16:
A e-mail from AI support
_"
Hi Gilles,
From the picture in your last email, the development boards you received should be in different batches. The left side of the picture is the newer module (A1S_V2.3), and the right side is the previous version of the module (A1S_V2. 2)
And the two only have the difference in the version.
The hardware interface is different, the IIC of V2.3 is 32, 33, and the IIC of V2.2 is 18, 23

It is recommended to use the V2.3 version."_

**Here the V2.2 and V2.3 are the version of the ESP32 Module not the Audio Board**

But we can not detect or read the version of the A1S..... :-(

superjasiek, 2021-11-18:
This thread partially solved my problem about pin assignment  of audio codec in A1S Audio Kit v2.2 3378 series (i was trying to use it with esp32-squeezelite). Unfortunately can't figure out what pin is responsible for jack detection and what kind of configuration of dip switches should be. 
Any idea? thank's for replay in advance.

GillesLACAUD, 2021-11-18:
Try IO39 for the Headphone detection

superjasiek, 2021-11-18:
Already tried. But maybe it's a matter of software - that's why i need to know for sure what pin it is. Maybe someone has this version (3378) and compiled from sources something with jack detection?

lderani, 2021-11-19:
> Finally I found this solution for low output volume at earphone jack. Just modify ADF audio hall 8366 driver (esp-adf\components\audio_hal\driver\es8388\es8388.c) Find "es8388_set_voice_volume" [...]

Hi @tttin46 , I have a 2957 board, and to make the sound loud at the headphone jack I:

- changed the i2c pins to: 

        i2c_config->sda_io_num = GPIO_NUM_33;
        i2c_config->scl_io_num = GPIO_NUM_32;

and

        i2s_config->bck_io_num = GPIO_NUM_27;
        i2s_config->ws_io_num = GPIO_NUM_25;
        i2s_config->data_out_num = GPIO_NUM_26;
        i2s_config->data_in_num = GPIO_NUM_35;

as instructed on https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit for the boards with the ES8388 DAC. After that I  changed the es8338.c file like you  did, and added:

#include "es8388.h" at the include section of the example, and

es8388_set_voice_volume(100);

just before the #ifdefs to choose the file decoder type, and the music played at max volume on the headphone jack.

Thank you for your VERY useful solution! :-)

vanbwodonk, 2021-11-19:
Just if you guys need Arduino version es8388 for esp32-a1s, tested on 18/23 and 33/32 i2c version.
https://github.com/vanbwodonk/es8388arduino. I hope solved some of your problems. Still bad code,PR welcome.

lderani, 2021-11-20:
@superjasiek , I have the ver.2957 board and I managed to make the "pipeline_play_sdcard_muisc" example detect headphone insertion/removal. Maybe is the same on your version (3378). Here's what I did, if you wanna try:

[image]

Since I checked that the detection sleeve of the headphone is linked to IO pin 39, I just changed the #define for HEADPHONE_DETECT  on the file:

%ADF_PATH%\components\audio_board\lyrat_v4_3\board_def.h

from:
#define HEADPHONE_DETECT          GPIO_NUM_19 (original pin on LyraT board)
to
#define HEADPHONE_DETECT          GPIO_NUM_39 (pin on the A1S)

In the example I used, the detection is made on a ISR inside this file:

%ADF_PATH%\components\audio_hal\driver\es8388\headphone_detect.c

as soon as I recompiled the example with the changed #define, removal/insertion worked. 

I hope this helps you!

superjasiek, 2021-11-20:
Thanks for for the hint! I made some measurements and for model 3378 GPIO for jack detection is 39 but the voltage without jack plugged is about 3,12v.

VorkHeftruck, 2021-11-21:
@vanbwodonk thank you for sharing. Doesn’t look like bad code to me! Clear and clean coding. Will give feedback once I gave it a good run but from a first read through looks like a very useful driver and examples. Well done.

vijugo, 2021-11-24:
> I bought a "ESP32 Audio Kit v2.2 2957" and installed the toolchain as described in this readly, and the ADF (+IDF) from this repo. I can compile and flash all examples I tried so far (here, e.g. [...]

GillesLACAUD, 2021-11-24:
I think you have an ES8388 codec and not an AC101. AC101 is only for the old version.
Please read carefully all the post and i pretty sure you will find the solution

vijugo, 2021-11-24:
En la libreria https://github.com/schreibfaul1/es8388

Modifique la dirección del dispositivo de 0x1a a 0x10 en el archivo ES8388.cpp

#include <Arduino.h>
#include "ES8388.h"
#include <Wire.h>

**#define ES8388_ADDR 0x10**

Proceda a escribir este programa 

// the pin assignment matches the Olimex ADF board

#include "Arduino.h"
#include "WiFi.h"
#include "SPI.h"
#include "SD.h"
#include "FS.h"
#include "Wire.h"
#include "ES8388.h"  // https://github.com/maditnerd/es8388

#define SD_CS         13

// GPIOs for SPI
#define SPI_MOSI      15
#define SPI_MISO      2
#define SPI_SCK       14

// I2S GPIOs
#define I2S_SDOUT     26
#define I2S_BCLK      27
#define I2S_LRCK      25
#define I2S_MCLK       0

// I2C GPIOs
#define IIC_CLK       32
#define IIC_DATA      33

// Amplifier enable
#define GPIO_PA_EN                  GPIO_NUM_21
#define GPIO_SEL_PA_EN              GPIO_SEL_21

char ssid[] =     "XXXX";
char password[] = "XXXX";

int volume = 100;                            // 0...100

ES8388 es;
Audio audio;

//----------------------------------------------------------------------------------------------------------------------

void setup()
{
  Serial.begin(115200);
  Serial.println("\r\nReset");
  Serial.printf_P(PSTR("Free mem=%d\n"), ESP.getFreeHeap());

  SPI.begin(SPI_SCK, SPI_MISO, SPI_MOSI);
  SPI.setFrequency(1000000);

  SD.begin(SD_CS);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(100);
  }

  Serial.printf_P(PSTR("Connected\r\nRSSI: "));
  Serial.print(WiFi.RSSI());
  Serial.print(" IP: ");
  Serial.println(WiFi.localIP());

  Serial.printf("Connect to ES8388 codec... ");
  while (not es.begin(IIC_DATA, IIC_CLK))
  {
    Serial.printf("Failed!\n");
    delay(1000);
  }
  Serial.printf("OK\n");

  es.volume(ES8388::ES_MAIN, volume);
  es.volume(ES8388::ES_OUT1, volume);
  es.volume(ES8388::ES_OUT2, volume);
  es.mute(ES8388::ES_OUT1, false);
  es.mute(ES8388::ES_OUT2, false);
  es.mute(ES8388::ES_MAIN, false);

  // Enable amplifier
  pinMode(GPIO_PA_EN, OUTPUT);
  digitalWrite(GPIO_PA_EN, HIGH);

  audio.setPinout(I2S_BCLK, I2S_LRCK, I2S_SDOUT);
  audio.i2s_mclk_pin_select(I2S_MCLK);
  audio.setVolume(21); // 0...21

  audio.connecttohost("http://mp3channels.webradio.antenne.de:80/oldies-but-goldies");

  //audio.connecttoFS(SD, "01/001.mp3");
  //  audio.connecttospeech("Wenn die Hunde schlafen, kann der Wolf gut Schafe stehlen.", "de");
}
//----------------------------------------------------------------------------------------------------------------------
void loop()
{
  audio.loop();
}
//----------------------------------------------------------------------------------------------------------------------

// optional
void audio_info(const char *info) {
  Serial.print("info        "); Serial.println(info);
}
void audio_id3data(const char *info) { //id3 metadata
  Serial.print("id3data     "); Serial.println(info);
}
void audio_eof_mp3(const char *info) { //end of file
  Serial.print("eof_mp3     "); Serial.println(info);
}
void audio_showstation(const char *info) {
  Serial.print("station     "); Serial.println(info);
}
void audio_showstreaminfo(const char *info) {
  Serial.print("streaminfo  "); Serial.println(info);
}
void audio_showstreamtitle(const char *info) {
  Serial.print("streamtitle "); Serial.println(info);
}
void audio_bitrate(const char *info) {
  Serial.print("bitrate     "); Serial.println(info);
}
void audio_commercial(const char *info) { //duration in sec
  Serial.print("commercial  "); Serial.println(info);
}
void audio_icyurl(const char *info) { //homepage
  Serial.print("icyurl      "); Serial.println(info);
}
void audio_lasthost(const char *info) { //stream URL played
  Serial.print("lasthost    "); Serial.println(info);
}
void audio_eof_speech(const char *info) {
  Serial.print("eof_speech  "); Serial.println(info);
}

Modificar la linea de la dirección

vijugo, 2021-11-24:
Yo descargue está librería con un ESP32-A1S que internamente tiene un AC101

https://github.com/Yveaux/AC101

y funcionó correctamente

indotronic, 2021-12-05:
hi, i am get same issues using version 2.3, e3833 IC codec.
the headphone just a sounding

Dharm94, 2021-12-15:
Hi everyone:

First of thanks as this thread helped me a lot. But I'm getting one more error on the same topic.
[image]

Why this GPIO error is coming on every time when initialization has done. Nothing is getting recorded in wav file. 

Using ESP32 Audio Kit V2.2 (3378) & on the backside of A1s (2974)

mario44, 2021-12-15:
Witam, ostatnio kupiłem ESP32-A1S V2.2 3478
Potrzebuje opis pinów. Czy ktoś już dostał taką wersje?
[image]

superjasiek, 2021-12-15:
Well, it looks exactly the same as mine but mine is 3378. I use it for squeezelite player and found some pin configuration:

dac_config": "model=ES8388,bck=27,ws=25,do=26,sda=33,scl=32,i2c=16"
spi_config": "mosi=23,clk=18,dc=5,host=1"
set_GPIO": "21=amp,22=green:0,39=jack:0",

mario44, 2021-12-15:
@xuhongv Did you have any idea about the 3478 boards ?

mario44, 2021-12-15:
@superjasiek Thank you very much, I will try!
Could you please define more precisely what it is:
i2c = 16
dc = 5
host = 1

GillesLACAUD, 2021-12-15:
The version on the **ESP32 Audio Kit** is not important. The difference is in the **ESP32-A1S module**.
The only way to find the correct mapping (for I2c codec) is to **scan the I2C bus** and find the correct I2C adress (AC101 or ES8388) and the correct I2C mapping for ES8388 (Lyra Version) or ES8388 (Last version).
The correct (and the last version) is:
Board with ES8388 last version map at I2C pin 32/33

GillesLACAUD, 2021-12-15:
Did someone has some trouble with the TP4065 ?....When the component is in charging mode (only red led on) I can hear some noise and whistle on the audio output.
When the status is Ok (Green led On) the noise disappear.
I have find a workaround....I pull the CE pin for this component

mario44, 2021-12-15:
My project includes: Internet streaming playback and I2C LCD 1602 display. The problem is that the board sometimes restarts by itself. The es8388 audio codec uses the pins: (IIC_CLK 32, IIC_DATA 33), while the LCD display uses the pins: (IIC_DATA 21, IIC_CLK 23). I am looking for a complete description of all pins for this version 3478.
Below are the pins that I have marked in my code:
#define SD_CS 13

#define SPI_MOSI 15
#define SPI_MISO 2
#define SPI_SCK 14

#define I2S_DOUT 26
#define I2S_BCLK 27
#define I2S_LRC 25
#define I2S_MCLK 0
#define I2S_DIN 35

#define ES8388_IIC_CLK 32
#define ES8388_IIC_DATA 33

#define GPIO_PA_EN 21

#define LED_BUILTIN 22
#define KEY6 5
#define KEY5 18
#define KEY4 23
#define KEY3 19

RitikDutta, 2021-12-21:
I'm able to play MP3 songs from sd using examples but unable to record anything. i also have the same issue does anyone got the  solution??

[14 more replies]