Ai-Thinker-Open/ESP32-A1S-AudioKit #35: Please integrate board into platformio's board database.

cdluv, 2022-05-02:
Can't see this board in the PlatformIO plugin for VS Code.     I've had the board laying around for a couple of years, and because of the barrier to entry (AKA PITA) getting this board going, I've prioritised other projects in the meantime.

Would love to continue to work on this board when it's simpler to play with.

Raised this issue with Platform IO too, and it's been hanging around for nearly 18 months.  https://github.com/platformio/platform-espressif32/issues/457

I hope that you consider improving community support.  It's good for business, and customers alike!

bmckay959, 2022-05-03:
Yes, please! Thanks @cdluv for bringing it up here.   

@xuhongv How realistic is it for this to be added? I bought 10 of the boards for a project and now having to send them back if I can't get them running.

Thanks for your feedback

bmckay959, 2022-05-05:
@xuhongv ,  can you please give an update here? Even if the answer is no, I will just need to return the device.

Is there a generic board that I can use in the meantime?

cdluv, 2022-05-20:
#tumbleweed

pschatzmann, 2022-05-30:
I suggest that you try with PlatformIO Arduino and the following libraries:
- https://github.com/pschatzmann/arduino-audio-tools
- https://github.com/pschatzmann/arduino-audiokit

In the Wiki there is a description how to use it with PlatformIO