Ai-Thinker-Open/ESP32-A1S-AudioKit #42: Correct pins for V2.2 2957 board

SerdarAbali, 2023-06-19:
I decided to skip [this](https://github.com/Ai-Thinker-Open/ESP32-A1S-AudioKit/issues/26) since it has too many irrelevant responses and super difficult to find a proper solution. 
Has anyone fixed this problem with the V2.2 2957 board?

E (1026) AUDIO_HAL: codec init failed!
E (1026) AUDIO_BOARD: /home/test/esp/ESP32-A1S-AudioKit/components/audio_board/ai_thinker_audio_kit_v2_2/board.c:55 (audio_board_codec_init): Got NULL Pointer
E (1036) AUDIO_HAL: audio_hal handle is null

Thanks in advance.

trombik, 2023-07-10:
here is [my forked `esp-adf`](https://github.com/trombik/esp-adf/tree/ai-thinker-esp32-a1s-audio-kit/) that supports `ESP32-A1S` with `ES8388`. see the details at [README.md](https://github.com/trombik/esp-adf/tree/ai-thinker-esp32-a1s-audio-kit/components/audio_board/ai_thinker_audio_kit).

by the way, the zip files @lderani mentioned in the above comment lack some files, and do not work.

lderani, 2023-07-10:
Well, dunno why, but it works for me. Right now I'm compiling a project with IDF+ADF that uses wifi, sdcard, a serial control interface, web server, mic recording and  media playing with volume controls using my mods. and it works fine :-) . But hey, the more the merrier! Thanks for putting your mods on a proper new fork, and good luck with your projects! 👍

trombik, 2023-07-24:
instead of forking, i made a `esp-idf` component: https://github.com/trombik/esp-adf-component-ai-thinker-esp32-a1s