Ai-Thinker-Open/ESP32-A1S-AudioKit #44: Only REC and MODE buttons working on Audio Kit V2.2 A247 - A1S module B248 (possibly 8248)

osfog, 2023-11-20:
We are  only able to get REC and MODE buttons to work (buttons 1 and 3)

Confirmed by running the example:  
esp-adf/examples/checks/check_board_buttons/

We tried both with the apparently faulty definition you get when following the official instructions (basing builds on Lyra 4_3 board definition )

Have also tried with setting input mode and pull-ups on the relevant pins.

```
#define FUNC_BUTTON_EN            (1)
#define INPUT_KEY_NUM             6
#define BUTTON_REC_ID             GPIO_NUM_36
#define BUTTON_MODE_ID            GPIO_NUM_39
#define BUTTON_SET_ID             TOUCH_PAD_NUM9
#define BUTTON_PLAY_ID            TOUCH_PAD_NUM8
#define BUTTON_VOLUP_ID           TOUCH_PAD_NUM7
#define BUTTON_VOLDOWN_ID         TOUCH_PAD_NUM4
```

And the more reasonable:

```
#define BUTTON_REC_ID             GPIO_NUM_36   //KEY1
#define BUTTON_MODE_ID            GPIO_NUM_19   //KEY3
#define BUTTON_SET_ID             GPIO_NUM_13   //KEY2
#define BUTTON_PLAY_ID            GPIO_NUM_23   //KEY4
#define BUTTON_VOLUP_ID           GPIO_NUM_18   //KEY5
#define BUTTON_VOLDOWN_ID         GPIO_NUM_5    //KEY6
```

(with corresponding modificaitons to the key setup:
```
#define ESP32_A1S_AUDIO_KEY1    (1ULL << get_input_rec_id())
#define ESP32_A1S_AUDIO_KEY2    (1ULL << get_input_set_id())
#define ESP32_A1S_AUDIO_KEY3    (1ULL << get_input_mode_id())
#define ESP32_A1S_AUDIO_KEY4    (1ULL << get_input_play_id())   
#define ESP32_A1S_AUDIO_KEY5    (1ULL << get_input_volup_id())
#define ESP32_A1S_AUDIO_KEY6    (1ULL << get_input_voldown_id())

esp_err_t audio_board_key_init(esp_periph_set_handle_t set)
{
    periph_button_cfg_t btn_cfg = {
        .gpio_mask =  ESP32_A1S_AUDIO_KEY1 | 
                      ESP32_A1S_AUDIO_KEY2 |
                      ESP32_A1S_AUDIO_KEY3 |
                      ESP32_A1S_AUDIO_KEY4 |
                      ESP32_A1S_AUDIO_KEY5 |
                      ESP32_A1S_AUDIO_KEY6 , //REC BTN & MODE BTN
    };
    ....
```

)

osfog, 2023-11-28:
Well the code that  had not included was where the culprit was ...  the INPUT_KEY_DEFAULT_INFO  types were set to PERIPH_ID_TOUCH  (inherited from the lyra board)

windtersharp, 2024-03-27:
> Well the code that had not included was where the culprit was ... the INPUT_KEY_DEFAULT_INFO types were set to PERIPH_ID_TOUCH (inherited from the lyra board)

Hello， V2.2 A247 is ac101 ?  my board is v2.2 A252